quartiq/thermostat-eem #27: AsmDelay delays are about 1/4 the time they should be

nkrackow, 2022-04-08:
Using `AsmDelay` from `asm_delay` form the asm-delay-0.9.0 crate with either the `delay_us()` or `delay_ms()` function leads to a delay that is about 1/4th the time specified. There are a couple of things I checked/don't understand:
- The clock frequency going into the new function (`ccdr.clocks.c_ck().0`) is as expected at 400 MHz
- If I just use the raw `cortex_m::asm::delay()` functions with 400_000 cycles I get a delay of 500 us instead of the 1 ms I expect.
- [here](https://github.com/copterust/asm-delay/blob/master/src/lib.rs#L26-L27) there is a division by two, which explains why `AsmDelay` is slower by a factor of two compared to the raw `cortex_m::asm::delay()` which it also makes use of

jordens, 2022-04-08:
Iirc precisely because of dual issue etc, the cortex-m delay was made slower to guarantee the delay as a lower bound. Asm_delay might well be deprecated. recommend looking around in the ecosystem.

jordens, 2022-04-08:
c.f. https://github.com/rust-embedded/cortex-m/issues/236 
https://docs.rs/cortex-m/0.7.4/cortex_m/asm/fn.delay.html

nkrackow, 2022-04-11:
Interesting. I'll have a look what other options there are.

jordens, 2022-04-11:
Could you check and file issues iff this also affects stabilizer/thermostat-mqtt/booster?

nkrackow, 2022-04-11:
> Could you check and file issues iff this also affects stabilizer/thermostat-mqtt/booster?

Good point. Stabilizer yes but I think booster could be fine since it has the M4 and [here](https://github.com/quartiq/booster/blob/46ed6c823aa10059af6e9035068af54bb64c72b5/src/hardware/delay.rs#L11-L12) the delay is implemented without that division by two. Do you think that is fine then @ryan-summers?

ryan-summers, 2022-04-11:
Booster doesn't use `asm_delay`, so we should be good on that front

nkrackow, 2022-04-11:
I have now implemented a custom delay just like in [booster](https://github.com/quartiq/booster/blob/develop/src/hardware/delay.rs). Do we think that's a reasonable solution or could you point me to a better implementation from the ecosystem?

closed 2022-04-13 by nkrackow