quartiq/thermostat-eem #28: ADC communication probing

nkrackow, 2022-04-08:
I'll just collect a few observations of the ADC communication here.

Reading out all ADCs (that finish sampling at the same time) with our staggered interrupt routine and a 12.5 MHz SPI clock takes about 130 - 140 us. Blue is CS of the first ADC, which is low most of the time until it's data is read out and the other ADCs are selected. Yellow is the SPI dout signal, first going low to signal new data and trigger the interrupt and the giving back the data from the ADCs.
[image]

jordens, 2022-04-08:
Reason for the long time between reads?

nkrackow, 2022-04-11:
No, that is `--debug`. Good point.

nkrackow, 2022-04-11:
[image]
Here you can see that the next ADC that is selected immediately pulls dout low, triggering the interrupt. But then it takes quite a long time for the CPU to handle it. I think it actually fully completes the last interrupt with all the prints before handling the next. I'll look into how to use rtic such that a new interrupt can interrupt the currently running one.

The other thing is that it takes about 2 us for the CPU to deselect the current ADC and then 4 us to look up which ADC comes next and select it. This is `--debug`. With `--release` things are a bit faster (I think mostly the prints) but the same behavior.

jordens, 2022-04-11:
You are likely just measuring the speed of formatting.

nkrackow, 2022-04-11:
Ok, so it is not possible with rtic to pre-empt a hardware interrupt task with itself. But I don't think that's a problem. It's not like things could go wrong here when each interrupt is first completed while the new has already triggered. And we also don't gain any savings in CPU time since we have to perform the same operations either way.

nkrackow, 2022-04-11:
[image]
With `--release` and no prints things are very fast now. Also I have seen that every reading cycle looks exactly the same without any timing jitter if I give the isr a higher priority than other tasks. If it is the same there is some jitter.

jordens, 2022-04-11:
As mentioned before, the next interrupt will just pend and chain right away.

closed 2022-04-13 by nkrackow