quartiq/thermostat-eem #30: adc code to temperature conversion quantization

jordens, 2022-04-17:
https://github.com/quartiq/thermostat-eem/blob/47a7ec559bcc6c1f3f82ffa33cda836e494ec287/src/hardware/adc.rs#L48

quantization with f32 is 31 µK, 24 bit LSB is 5.5 µK
f64 is desirable (conversion quantization 10 dB below ADC noise) if the RMS noise is less than 5 LSB (`31 µK/5.5 µK/sqrt(12)/3`)

nkrackow, 2022-04-19:
I find this is not trivial at all.. After some thinking and experiments I can confirm your statement: https://colab.research.google.com/drive/1lMaYTT44xk7N5msobnKhWJ6QpQqIOevv?usp=sharing
But I don't yet fully grasp where the 31 uK step size arises from. I had different theories and I think there are several arithmetic nuances at play during the various computation steps.
I would say it's best to switch to f64 in any case to not limit ourselves here?

jordens, 2022-04-19:
Yes. Not trivial at all. I think this is consistent and hand waveable with 31 uK/300 K ~ epsilon of f32.
Yes. I'd additionally `impl From<AdcCode> for f32`. Then you can easily switch.

PR #32 (2022-04-19): f64 temperature calculation. Casting of constants happens at compile time.
resolves #30

Telemetry in f32 due to #31
commit 8c5200f8 (2022-04-19): f64 temperature calculation. Casting of constants happens at compile time. resolves #30