quartiq/thermostat-eem #38: FW v0.1 measurements

nkrackow, 2022-05-12:
- For 10K ref in the middle see if PWM/TEC output on all out channels makes a difference
- good data of 10K ref 1 kHz 1M samples -> log spectrum 
   - standard config
   - single ended (Ain- - REF-) performance 
   - with long cable (shorter timespan)
- absolute difference of 10K with/without cable 
- transfer function from NTC to TEC
- common mode rejection. inject into shorted NTC input.
- TEC output voltage noise spectrum with TEC connected and driven. Measure to gnd. (500 kHz vs. 1 MHz switchable with gpio in FW)
- probe and disturb around the board
- investigate header board temperature susceptibility

nkrackow, 2022-05-16:
## 10K ref data

### Setup
- [Vishay 10K reference resistor](https://www.digikey.de/de/products/detail/vishay-foil-resistors-division-of-vishay-precision-group/Y145310K0000V9L/2609894) 
- ADC output data rate set to 1007 Hz -> 503.5 Hz sample rate per Thermostat channel, 50% of time sampling each channel.
- Sinc5Sinc1 ADC filter, no postfilter.
- Room temperature headerboard directly connected to Thermostat, 10K REF leads directly in Molex campling connector block.

### Data
[raw.zip](https://github.com/quartiq/thermostat-eem/files/8699736/raw.zip)
[image]

### Analysis
- Standard deviation over full bandwidth (250 Hz): 9.679 LSB, 51.4368 uK
- RMS noise 0.01 Hz - 1 Hz: 0.6323 LSB/Hz,  3.3601 uK/Hz

jordens, 2022-05-16:
@nkrackow do keep an eye on https://github.com/sinara-hw/Thermostat/issues/46 and compare

nkrackow, 2022-05-16:
## 10K ref with cable

### Setup
- [Vishay 10K reference resistor](https://www.digikey.de/de/products/detail/vishay-foil-resistors-division-of-vishay-precision-group/Y145310K0000V9L/2609894) 
- ADC output data rate set to 1007 Hz -> 503.5 Hz sample rate per Thermostat channel, 50% of time sampling each channel.
- Sinc5Sinc1 ADC filter, no postfilter.
- Room temperature headerboard onnected to Thermostat **via a ~0.5 m 24 AWG D-SUB cable**, 10K REF leads directly in Molex campling connector block.

### Data
[raw_cable.txt](https://github.com/quartiq/thermostat-eem/files/8700086/raw_cable.txt)
[image]

### Analysis
- Standard deviation over full bandwidth (250 Hz): 9.584 LSB, 50.9279 uK
- RMS noise 0.01 Hz - 1 Hz: 0.6199 LSB,  3.2943 uK
- Mean value difference to 10K ref without cable: +16.8 LSB

jordens, 2022-05-16:
Note to self: two channels: the noise bandwidth is 500 Hz

nkrackow, 2022-05-16:
## TEC output crosstalk

There is a slight, likely thermal effect of the TEC output back to the adc channels. Same settings as with "10K ref data", 3A TEC current when enabled. TEC output channel 3 traces seems to be the closest to all of the adc inputs on the layout.

### Input channel 5, output channel 3, headboard directly at Thermostat:
[image]

### Input channel 3, output channel 3, headboard directly at Thermostat:
[image]

### Input channel 3, output channel 3, ~0.5 m cable to Thermostat:
[image]

jordens, 2022-05-16:
How large is it?

nkrackow, 2022-05-17:
> How large is it?
I annotated it in the comment.

nkrackow, 2022-05-17:
## Thermal disturbance of the boards

The following data was taken under the same conditions again. At three times I heated up some part of the boards with a hot air station.The times of disturbance are annotated and the corresponding thermal image and maximum deviation is shown below. DIfference is with respect to the start of the time series.
[image]

### Base image of the board sampling data on all channels, all outputs disabled
[image]

### [1] ADCs and surrounding circuitry to about 120 °C
[image]
Max. deviation: 350 LSB

### [2] D-SUB connector on the board side to about 120 °C
[image]
Max. deviation:  -80 LSB

### [3] Headboard bottom side where the reference resistors are located to about 120 °C
ambient (before disturbance):
[image]
heated:
[image]
Max. deviation: 570 LSB

nkrackow, 2022-05-17:
## Aspirate onto the bottom of the headboard

By far the biggest disturbance I noticed is by aspirating onto the bottom of the headboard such that a little moisture condensates on the board. It is visible as tiny droplets (usual condensation) but those were difficult to photograph. The board temperature only changes by about 2 °C. 

I have aspirated only about the first 15% of the disturbance in the data. Then the deviations stays roughly constant until it rapidly goes back to the initial value. This is consistent with visibility of condensation on the board.

[image]
Max. deviation: 3440 LSB, about 0.02 °C

nkrackow, 2022-05-19:
##  Frequency response

For the frequency response and group delay measurements the temperature conversion step was substituted by just a constant conversion coefficient so that the input-output relation is linear over the full scale. All the other processing like IIR and current to dac_code conversion are still present. The group delay should therefore not be significantly effected since much of it is in the ADC/DAC.

The following frequency magnitude responses measurements were done with a 2 V peak to peak, 2.5 V offset stimulation between the positive differential ADC input and the negative reference. The negative ADC input is still pulled to the negative reference by a 5K reference resistor on the headboard. The output was taken over a 1 ohm load resistor connected at the TEC output. The IIR only had the first b coefficient ("proportional gain") set to a value such that the output is between about 0.3 and 0.8 A. First input channel (ADC0 ch 0) and first output channel were used.

### Setup
[image]

### Frequency response 
[image]
[image]
First plot is from HP89410A with 100x averaging, second from an Analog Discovery. The ADC first order CIC response (with 1007 Hz effective rate) can clearly be seen. Since only half the samples go to each ADC channel, the nyquist rate for one channel is about 250 Hz. In the analog discovery plot (wich is not averaged) you can see some beating effects around the nyquist rate.

### Group delay
[image]
Group delay is about 3.2 ms.

nkrackow, 2022-05-19:
## TEC driver output spectrum
[image]
The driver switching frequency can be seen at bout 580 kHz.

nkrackow, 2022-05-19:
## DAC time constant
[image]
Output time constant is about 210 us. The 1000 kHz lowpass installed between DAC and TEC driver should have about 160 us.

jordens, 2022-05-19:
The analog discovery has a differential input with floating shield?
Ans why is the second lobe now -23 dB? Is that understood from the filter/decimation chains?
Some quantitative numbers would be useful. It's ok if they have a errorbar (then estimate that errorbar).

* TEC current thermal crosstalk magnitude (LSB/K)
* TEC current direct crosstalk (or upper limit if not detected)
* output ripple (A pk-pk worst case)
* output 580 kHz amplitude (A pk-pk worst case)
* a reasonable magnitude of the tempco of the board (LSB/K)
* Common mode voltage ripple (or upper limit)

nkrackow, 2022-05-19:
> The analog discovery has a differential input with floating shield? 

Differental yes, floating no. You can measure up to 50 V wrt. ground differentially.

>Ans why is the second lobe now -23 dB? Is that understood from the filter/decimation chains? 

I think it was the same yesterday. Also the one plot is from yesterday and there it is about that too.

>Some quantitative numbers would be useful. It's ok if they have a errorbar (then estimate that errorbar).
> 
> * TEC current thermal crosstalk magnitude (LSB/K)

For which channel/configuration is that? Just the one channel that I suspect to be the most effected? Should I assume linear correspondence between temperature of the TEC driver chip and the output deviation?

> * TEC current direct crosstalk (or upper limit if not detected)

TODO.

> * output ripple (A pk-pk worst case)

I am not capable of determining this accurately. I see significant switching noise at about 68 kHz which must come from some power supply here somewhere. I tried many different ways of grounding but I can not get rid of it. I know that it is not trivial to accurately measure power supply ripple. This is with 3A output. The switching noise is also there when the output is off.
[image]

If I try to look through this switching noise this looks like below 10 mApp noise. 
[image]

> * output 580 kHz amplitude (A pk-pk worst case)

About 200 uApp right at this frequency. Too small to see in the time domain. The y scale in the spectrum is in absolute V peak to peak (with a 1 ohm load).

[image]

> * a reasonable magnitude of the tempco of the board (LSB/K)

TODO. That would be for the whole board uniformly?

> * Common mode voltage ripple (or upper limit)

Same problem as with measuring the output ripple. Would you measure this from the load resistor lower side to ground?

jordens, 2022-05-19:
Cool!

jordens, 2022-05-19:
If you expect the ripple to be at 500 kHz, shouldn't you choose a horizontal scale in the ~µs/div range instead of ms/div?

jordens, 2022-05-19:
RC mismatch maybe? how large would the capacitance or resistance mismatch need to be?
Oh. And your stimulus may also be affecting/driving the REF+ channel (finite impedance). Maybe try removing the broken 5k ref resistor and using just that channel (should not need to touch anything else).

nkrackow, 2022-05-19:
> If you expect the ripple to be at 500 kHz, shouldn't you choose a horizontal scale in the ~µs/div range instead of ms/div?

That was just the timescale where the most noise was apparent. As stated above, the 570 kHz aren't visible in the time domain.

nkrackow, 2022-05-19:
> RC mismatch maybe? how large would the capacitance or resistance mismatch need to be? Oh. And your stimulus may also be affecting/driving the REF+ channel (finite impedance). Maybe try removing the broken 5k ref resistor and using just that channel (should not need to touch anything else).

Good point, currently trying that.

jordens, 2022-05-19:
But -90 dB at 50 Hz also sounds good.

nkrackow, 2022-05-19:
[image]
This is on the input channel with the broken 5K that i just removed. 
The other difference is that I changed the offset in the ADC sucht that I could apply +10 mV offset to the differential stimulus, still 10 mV amplitude. 

Now it's even a bit better but still some highpass dynamic going on. Maybe even more gain in the IIR would help to see more but one measurement takes about 20 min..

jordens, 2022-05-19:
Ah. You used the same stimulus and same IIR gain for both measurements? There is no need for that.
In your figure you are limited by your measurement noise for CM < 10 Hz and both CM and DM > 1 kHz. The data doesn't say much in those regions.
Why not large stimulus (~V) in both cases, and setting the IIR gain so that you get maximum signal (at the DAC) in each case? Then just correct the transfer functions for the difference in gain.
I presume this is still using the DAC and the Analog Disco board or are you just using the ADC samples and RTT?

jordens, 2022-05-19:
Anyway. This is good enough already. Nothing to worry about.
You can stop now ;)

nkrackow, 2022-05-19:
> Ah. You used the same stimulus and same IIR gain for both measurements? 

No, like in the first CMRR measurements 2 V offset 1 V amplitude for common mode stimulus. Just the differential stimulus changed to 10 mV offset, 10 mV amplitude.

jordens, 2022-05-19:
Increasing the IIR coefficient for CMRR is what raises the CM signal out of the output stage quantization, ripple, and measurement noise. Just give it 40-80 dB more gain than DM.
And why was the DM amplitude only 10 mV? Don't you want large amplitude there as well to not be limited?

nkrackow, 2022-05-20:
> And why was the DM amplitude only 10 mV?

Because that already makes about 0.5 A peak to peak output. I guess I could have increased it by a factor of six but then I'd have to hit exactly mid scale with the offset and that's not much in dB..

jordens, 2022-05-20:
For DM the gain should be low. Otherwise with low stimulus you'll be limited by the noise of your stimulus.

jordens, 2022-05-20:
Shows 50 dB more cmrr above 10 kHz. Bow you are only limited by SNR for DM. Anyway. Goot enough. Let's move on.

nkrackow, 2022-05-20:
Not sure about the meaningfulness of the data above 2-3 kHz. Even with less gain for DM I'd still have to counter the ADC filter roll-off. I guess you might see the high-pass trend continue to a bit higher up then.

jordens, 2022-05-20:
Exactly. That's what you want to see, not the noise floor of your Stimulus.

jordens, 2022-05-24:
@nkrackow it would be great to summarize the measurements in numbers in the style of a datasheet where possible.
AFAICT a couple take home messages already:

* Limited by ADC white noise from 10 mHz to > 250 Hz
* Beats the legacy Thermostat in white noise and latency significantly (legacy: 3.1 LSB rms @ 27 ODR, this: 10 LSB rms @ 1007 ODR) even without the 50 Hz notches.
* CMRR > 100 dB for f <= 50 Hz
* ...

jordens, 2024-02-12:
nothing to be done here

closed 2024-02-12 by jordens