quartiq/thermostat-eem #52: More flexible ADC setups (mix of single-ended and differential ratiometric control, sequencing of those flexible setups)

nkrackow, 2022-08-22:


jordens, 2022-11-07:
It would be really interesting to see the raw data for the single ended setup. For that with the current afe just change the ADC setup to 4 single ended channels everywhere (plus the impact on sequencing, data arrays, and timing), short the NTC and measure 5k ref vs the other 5k ref ratiometric.

jordens, 2022-11-07:
Ah I see we have the data more or less in https://github.com/quartiq/thermostat-eem/issues/38#issuecomment-1127683289 with slightly different conditions. But that should not matter.
Summary is that there is negligible benefit of the balanced measurement in this case.
The next step would be A firmware version that only supports 16 single ended sensors (i.e. 5 wire measurement with 2 excitation, and 2 reference per adc plus one signal sense wire per sensor).